30350n/pcb2blender #79: [Question]

set-soft, 2024-11-06:
Hi!
I'm trying to create fresh docker images for KiBot using the last release and Blender 4.2.3 LTS.
I want to programmatically install the addon.
The code for Blender 3.5 doesn't work:

```python
import bpy

# Register the addon and enable it
bpy.context.preferences.filepaths.script_directory = '/usr/bin/4.2/scripts/'
bpy.ops.preferences.addon_install(filepath='./pcb2blender_importer_2-14.zip', target='PREFS')
bpy.ops.preferences.addon_enable(module='pcb2blender_importer')
bpy.ops.wm.save_userpref()
```

It complains on the first assignment. I couldn't find much information about how to do it with the new API, and ChatGPT suggests using:

```python
import bpy

addon_path = "/var/pcb3d_importer.zip"
addon_name = "pcb3d_importer"
bpy.ops.preferences.addon_install(filepath=addon_path)
bpy.ops.preferences.addon_enable(module=addon_name)
bpy.ops.wm.save_userpref()
# Verify the addon is enabled
addon = bpy.context.preferences.addons.get(addon_name)
if addon:
    print(f"Addon '{addon_name}' installed and enabled successfully!")
else:
    print(f"Failed to enable '{addon_name}'.")
```

Which kind of work ... 
I'm installing the dependencies by hand, so I don't need the magic that installs them.
The problems are:

1. This method needs the old zip layout, the addon must be inside a folder, not at the root of the zip file
2. The __init__.py must contain bl_info, otherwise is installed but fails to be detected correctly

I'm repacking the zip file and patching it with the bl_info (adapter from older releases), it works, but I wonder if you know a better way that doesn't need this manipulation.
I

set-soft, 2024-11-13:
> You can install extensions via the [Blender CLI](https://docs.blender.org/manual/en/latest/advanced/command_line/extension_arguments.html#subcommand-install), either by name from the official repository (recommended) or directly from a file.
> 
> `blender --command extension install --enable pcb3d-importer` should do the trick I'm pretty sure.

Thanks for the pointer, I had to use:

```
blender --online-mode --command extension sync 
blender --online-mode --command extension install --enable pcb3d_importer
```

Note the change in the name

30350n, 2024-11-13:
Ah yeah, that makes sense. You need `--online-mode` and `--command extension sync` because you are probably on a fresh install.

Is everything working now for you? Any issues with dependencies?

30350n, 2024-11-13:
Ah okay, well seems like they fixed it already at least. (Should be part of 4.3, as well as the next 4.2 bugfix release).

set-soft, 2024-11-13:
I manually installed the dependencies and I got a little bit farther, now I must figure out how to call importer. 
In previous versions I used `bpy.ops.pcb2blender.import_pcb3d` but now `bpy.ops.pcb3d_importer.import_pcb3d` says:

```
Traceback (most recent call last):
  File "/home/salvador/0Data/Eccosur/kibot/kibot/blender_scripts/blender_export.py", line 443, in <module>
    main()
  File "/home/salvador/0Data/Eccosur/kibot/kibot/blender_scripts/blender_export.py", line 406, in main
    bpy.ops.pcb3d_importer.import_pcb3d(**ops)
  File "/usr/bin/4.2/scripts/modules/bpy/ops.py", line 109, in __call__
    ret = _op_call(self.idname_py(), kw)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: Calling operator "bpy.ops.pcb3d_importer.import_pcb3d" error, could not be found
```

30350n, 2024-11-13:
Yeah, it's in [here](https://projects.blender.org/blender/blender/issues/124452).
[image]

30350n, 2024-11-13:
> I manually installed the dependencies and I got a little bit farther, now I must figure out how to call importer.
> In previous versions I used `bpy.ops.pcb2blender.import_pcb3d`

That should still work.

set-soft, 2024-11-14:
Ok, it works with the old name.

I'm installing the wheels manually to workaround the 4.2.3 bug.

Thanks!

closed 2024-11-14 by set-soft

30350n, 2024-11-22:
4.2.4 and 4.3 are out now btw. 4.2.4 should work flawlessly, while 4.3 is not supported yet.

set-soft, 2025-01-14:
Just a note: when I tried 4.2.3 and v2.16 the name changed from "pcb3d_importer" to "bl_ext.blender_org.pcb3d_importer"

set-soft, 2025-02-24:
Hi @30350n !
The name change is quite annoying ... if I install the plugin at system level the name is now `bl_ext.system.pcb3d_importer` do you know a way to make it uniform? I'm currently trying

-  pcb2blender_importer  (the original name)
-  pcb3d_importer (worked in 2.14)
-  bl_ext.blender_org.pcb3d_importer (needed by 2.16)
-  bl_ext.system.pcb3d_importer (needed by 2.16 at system level)

And people using stable KiBot is complaining: https://github.com/INTI-CMNB/KiBot/issues/781

30350n, 2025-03-07:
Yeah, I had to change the name to get the extension into the [Blender Extensions Platform](https://extensions.blender.org/) because they don't allow addon names to contain "Blender". Don't think there's a unified way to handle this. You could either somehow enforce specific, known versions or `try, except` the different variants I guess.

set-soft, 2025-03-08:
Ok, don't worry, I'll patch the docker images. The next release will include the new script that tries the various names.
The only bizarre thing is that I must try a name for the globally installed plug-in (bl_ext.system.pcb3d_importer) and another name for the user level installed plug-in (bl_ext.blender_org.pcb3d_importer)

30350n, 2025-03-08:
Yeah, that depends on which "repository" the extension is installed in I guess.

[image]

set-soft, 2025-03-10:
Yes, and I guess this is to allow people to have more than one version installed. I.e. the system has v1.0.0 and a user wants to install v1.0.1 and doesn't have administrator rights.
But is quite complex from the Python side.