30350n/pcb2blender #82: I am not able to import PCB with new version of blender plugin

goran-mahovlic, 2024-11-22:
Thank you for your project it is really amazing!

Last week I was successful in importing PCB that was exported from Linux KiCad, today I have updated plugin in blender as new one was available, and I am no more able to do the import

Error I am getting - even with older PCB that I was able to import last week

[image]

Version of KiCad plugin ( it is same version as last time )

[image]

KiCad version

[image]

Updated Blender plugin version that is not longer working on my side 

[image]

Best,
Goran

30350n, 2024-11-22:
Thanks!

You are using Blender 4.3 which came out 3 days ago and isn't supported yet.

30350n, 2024-11-22:
If it just closes that's probably a crash and iirc crash logs aren't exactly useful. I think this can sometimes be caused by custom models sometimes, you could try importing without 3D component models (there's an option for that in the import UI).

(You can also share the `.pcb3d` file if you want, then I could take a quick look at it.)

goran-mahovlic, 2024-11-23:
I think I have low memory for this board, as I have lots of components.
I mainly work on open source boards, but unfortunately the one that fails is not open, so I am not able to share it.

When I get one of the open to fail I will let you know!

But here is the one that is open and I was able to import and render!

[image]

Again, thank you on your support and making export/import easy!

ziomau10, 2024-12-04:
Hi I also am facing this issue.
I can attach the pcb3d file of a board that gives similat traceback on a missing shadow_method as above.
Actually after the error I have in the viewport the PCB and the front traces and the front PADS.
No traces and no pads of the bottom side and no components.
Hope this helps you to find the problem.
[Sense.zip](https://github.com/user-attachments/files/18012704/Sense.zip)

Anyway great job!

30350n, 2024-12-05:
Blender 4.3 is supported now with [v2.16](https://github.com/30350n/pcb2blender/releases/tag/v2.16-k8.0-b4.2lts).

closed 2024-12-05 by 30350n