30350n/pcb2blender #83: Problems when mapping images to rotating PCB

Bazmundi, 2024-11-27:
A short clip of the problem [here](https://www.youtube.com/watch?v=Nk4LAW4T8AE).

Basically, all I did was break into the geo nodes at the end to put a transform inside a simulation region and was pumping time to the rotational coordinates.  

[image]

Likely the simplest and least useful approach, but I was only trying to create a endless loop of the board (eventually then as an animated GIF) for a posting on me blog.  Don't let that distract you, its all about the mapping of the image and you can see it happens through the center.  I tried moving the board off the world center etc.

Because the board is so small I have camera at 50mm and clipping start 0.000001 m so I might still play with scaling up the board etc. At the mo, the board's in as it's actual dimensions (so about size of a regular postage stamp, you remember those?)

Any thoughts please?

30350n, 2024-11-27:
> Basically, all I did was break into the geo nodes

That's the whole problem. You are directly modifying the geometry (instead of rotating the **object**). I'm not sure if what you are trying to do is possible with geometry nodes tbh, something like the old [animation nodes](https://github.com/JacquesLucke/animation_nodes) system would be a better fit here imo.

(This is "caused" by the material node tree I'm using, because it uses the Z coordinate of the geometry to map the textures to the correct side of the board. So if you are rotating the geometry, what you are getting is the expected behavior.)

It does not have anything to do with scaling. As you already noticed, the board is **correctly** scaled, this is generally helpful if you are working with other assets/reference material.

30350n, 2024-11-27:
Sounds good, thanks!

closed 2024-12-05 by 30350n