30350n/pcb2blender #84: [Question] Is line art rendering possible ?

hpopols, 2024-12-07:
Hi,

First of all, thank you for this fantastic plugin.

I am quite inexperienced with blender but I am able to properly use your plugin to export/import the PCB and manage to produce renderings.

My question is simple: I would love to produce line art render of the PCB. Is this something possible with the model exported from freecad or not ?

To illustrate what I need, I found the following image. It would be perfect if I could include including tracks/copper planes small outlines as well:
[image]

I searched and tryed many things without much success (due to my inexperience). The closest thing I found was : https://blender.stackexchange.com/a/289608
[image]

30350n, 2024-12-10:
Hi! I actually really like this idea so I've been experimenting around with it for a couple days and got quite promising results:

[image]

To achive this you have to:
1. Import a `.pcb3d` file using the `3D (deprecated)` PCB Material setting
2. Scale up the resulting model by a factor of 100
  (I generally advise against doing any kind of scaling, but in this case it's required, because otherwise the lines are way too thick)
3. Add a Grease Pencil object, using the "Collection Line Art" preset
4. Move the Grease Pencil Object to the Scene Collection
5. Ensure the Grease Pencil Object's Line Art Modifier uses the Collection the PCB is in and holdout the Collection
[image]
6. Add a Camera and position it so that the PCB is in frame
7. Render and optionally use the Compositor node setup you shared above to tweak the colors

30350n, 2024-12-10:
This only shows silkscreen like that, because the traces are inside the PCB model. If you want to show them as well, you can select them and raise their Z coordinate a bit (this is easiest done by selecting them by material like this:)
[image]

30350n, 2024-12-10:
Okay another thing (excuse my language, but this is FUCKING SICK):

Turns out you don't even have to do any rendering in Blender, but you can simply export the Grease Pencil object as SVG:

[image]

hpopols, 2024-12-11:
Hi Bobbe!

No need to apologize when it is actually pretty FUCKING SICK! ;)

Anyway, I played around last night and I did succeed to render one of my board and even export it to SVG.

I have some weird lines which are converted to grease lines:
[image]

But this can be fixed after some manual SVG cleanup via inkscape:
[image]

Is there a way to avoid this manual step ? (It would be nice but is definitely not a big deal)

However, I did not succeed to raise the tracks because I can find the "select button" as you shown:
[image]

Thank you very much for your support again,

Xavier

closed 2025-05-24 by 30350n