30350n/pcb2blender #85: Cannot load PCB

blackbit42, 2024-12-11:
After experiencing various issues (https://github.com/30350n/pcb2blender/issues/62, https://github.com/30350n/pcb2blender/issues/65), I've downloaded a blender binary build, with which `pcb2blender` works fine for some simple kicad projects of mine.
Though, when I try to load [MAUreene_8226](https://github.com/blackbit42/MAUreene_8226), blender stalls with no CPU utilization. After hitting ctrl-C, I get this:
```
^CTraceback (most recent call last):
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 84, in update_props
    self.inputs["Color"].default_value = (*srgb2lin(material.diffuse), 1.0)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/core.py", line 105, in srgb2lin
    def srgb2lin(color):
    
KeyboardInterrupt
File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 75, in update_props
^CTraceback (most recent call last):
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 84, in update_props
    self.inputs["Color"].default_value = (*srgb2lin(material.diffuse), 1.0)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/core.py", line 105, in srgb2lin
    def srgb2lin(color):
    
KeyboardInterrupt
File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 75, in update_props
^CTraceback (most recent call last):
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 157, in init
    self.init_node_tree(inputs, nodes, outputs)
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/custom_node_utils/__init__.py", line 52, in init_node_tree
    setup_node_tree(node_tree, nodes_def)
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/custom_node_utils/__init__.py", line 29, in setup_node_tree
    links.new(nodes[from_node].outputs[output_index], node.inputs[input_index])
KeyboardInterrupt
Traceback (most recent call last):
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 81, in update_props
    material = self.get_material()
               ^^^^^^^^^^^^^^^^^^^
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 165, in get_material
    color = "custom_" + rgb2hex(lin2srgb(self.inputs["Color"].default_value[:3]))
                                         ~~~~~~~~~~~^^^^^^^^^
KeyError: 'bpy_prop_collection[key]: key "Color" not found'
File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 75, in update_props
Traceback (most recent call last):
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 81, in update_props
    material = self.get_material()
               ^^^^^^^^^^^^^^^^^^^
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 165, in get_material
    color = "custom_" + rgb2hex(lin2srgb(self.inputs["Color"].default_value[:3]))
                                         ~~~~~~~~~~~^^^^^^^^^
KeyError: 'bpy_prop_collection[key]: key "Color" not found'
File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 75, in update_props
Traceback (most recent call last):
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 81, in update_props
    material = self.get_material()
               ^^^^^^^^^^^^^^^^^^^
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 165, in get_material
    color = "custom_" + rgb2hex(lin2srgb(self.inputs["Color"].default_value[:3]))
                                         ~~~~~~~~~~~^^^^^^^^^
KeyError: 'bpy_prop_collection[key]: key "Color" not found'
File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 75, in update_props
Traceback (most recent call last):
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/importer.py", line 298, in execute
    enhance_materials(self.new_materials)
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/materials.py", line 75, in enhance_materials
    mat4cad_mat.setup_node_tree(node_tree)
  File "/home/blackbit/.config/blender/4.3/extensions/blender_org/pcb3d_importer/mat4cad/blender/__init__.py", line 37, in setup_node_tree
    node_shader.inputs["Color"].default_value = (*srgb2lin(self.diffuse), 1.0)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^
KeyError: 'bpy_prop_collection[key]: key "Color" not found'
^CTraceback (most recent call last):
  File "/home/blackbit/Downloads/blender-4.3.0-linux-x64/4.3/scripts/startup/bl_ui/properties_paint_common.py", line 17, in poll
    @classmethod

KeyboardInterrupt
```

I guess a property of the project is incompatible with `pcb2blender`?

30350n, 2024-12-11:
Can you share the `.pcb3d` file?

Looking at the logs you shared, this might all be caused by the `KeyboardInterrupt` you triggered via `Ctrl + C`. There's a good chance that it just takes a while to import.

blackbit42, 2024-12-11:
[maureene.pcb3d.zip](https://github.com/user-attachments/files/18095593/maureene.pcb3d.zip)
That's the file I tried to load.

30350n, 2024-12-11:
Okay, testing it on my machine now.

Without even loading it though, there seem to be some relatively large models in there, which might take a while to load because the VRML from Blender I use is really poorly optimized.

[image]

So it might be a good idea to reduce detail on those, if possible. (I've created [free2ki](https://github.com/30350n/free2ki/) to export pcb2blender friendly models from FreeCAD to KiCad).

30350n, 2024-12-11:
Here's an example for the 4PDT_Relay, exported via `free2ki`:

[4PDT_Relay.zip](https://github.com/user-attachments/files/18095899/4PDT_Relay.zip)

blackbit42, 2024-12-11:
Oh, I wasn't aware you're into FreeCAD too.
This is super off-topic, I'll ask anyways (perhaps we should move this conversation somewhere else if it picks up pace...).
What I would like to achieve is to get various renders of a KiCAD PCB from KiCAD itself, FreeCAD and Blender, all with the same 'view'.
In FreeCAD there are some presets like [Std_ViewIsometric](https://wiki.freecad.org/Std_ViewIsometric) and [Std_ViewTrimetric](https://wiki.freecad.org/Std_ViewTrimetric).
I have a hard time figuring out how I can
* Determine the angles that those views use
* Do the same as FreeCAD in KiCAD and Blender

In case this is something you'd be interested in and willing to help with, please let me know so that we can chat elsewhere.

blackbit42, 2024-12-11:
This is strange.
I left the Blender window alone after kicking off the import, 2 hours ago or so.
Now it is responsive again, the import did _not_ happen and there are no messages on the shell whatsoever.

30350n, 2024-12-11:
> Oh, I wasn't aware you're into FreeCAD too. This is super off-topic, I'll ask anyways (perhaps we should move this conversation somewhere else if it picks up pace...). What I would like to achieve [...]

Hmm I also don't know a whole about this, but this sounds like it should be achievable using some basic linear algebra.

In Blender you can just directly set the cameras location, rotation, FOV, etc.:
[image]
[image]

For KiCAD, it seems like you can setup "Viewports" in the 3D Viewer to Save different camera positions/rotations. These get then saved in the `.kicad_pro` file (for some reason) as a 4x4 Matrix (note the messed up order though). So you could probably just insert your wanted transformation in there. From what I can tell it's not possible to set focal length/FOV though, so you'd have to find out which FOV KiCAD's camera uses and match it in Blender/FreeCAD.
[image]

For FreeCAD it should definitely be possible via Python and I've just given it a quick google search and also seen some Macros floating around. It should definitely be possible, but I would also have to do more research here.

(Of course all the coordinate systems, scales/units, etc. most likely won't match, so you'll have to find suitable conversions between the softwares.)

blackbit42, 2024-12-11:
Good suggestion, I'll do so, thank you.

blackbit42, 2024-12-11:
> > the import did not happen
> 
> Are you sure about that? The imported PCBs will appear relatively small (as they are scaled _correctly_ in regards to Blender's units).

You were right.
1. PCB was very small
2. the default cube blocked the view

Reality is that I have practically no clue of Blender and still have to find my feet.

blackbit42, 2024-12-12:
Progress...
I installed the free2ki mod into FreeCAD. That was easy.
The material modification and export too. I did that for one of the two large thingies in my KiCAD project for now.
Import into Blender is **fast**, rendering is **fast** too.
[image]
I've noticed though that I've lost the multi-coloring of the relay.
I guess I have to take this into account when doing the material thing in FreeCAD?

Anyways, I must pull what I export from FreeCAD into KiCAD instead of what is there now, save that and export the entire board again, which shall then go into Blender, yes?

blackbit42, 2024-12-12:
[image]
I just gave it a try. In principle the process works, I think.
* KiCAD is happy to accept the newly exported 3D models from FreeCAD
* I can export from KiCAD and import into Blender (the latter being reasonably fast in comparison to before)
The color issue persists, though - again - I think I know the reason for that.
What puzzles me though currently is the incorrect size of the relay and the transformer.

30350n, 2024-12-12:
> I've noticed though that I've lost the multi-coloring of the relay.
> I guess I have to take this into account when doing the material thing in FreeCAD?

Hmm that should be something wrong on your side then, just tested it and it's working fine for me. I setup the materials like this (as an example):
[image]

and they appear correctly in Blender:
[image]

> What puzzles me though currently is the incorrect size of the relay and the transformer.

`free2ki` (in contrast to other solutions, aka Step Up/default FreeCAD exports) actually scales the models **correctly** for KiCAD and Blender. My guess is that you directly replaced your old models (which had scaling applied) and didn't reset the scaling to 1.

Also, sidenote, you should probably not export that footprint/board mockup with the relay (to do so, select only the relay Part before hitting the free2ki export button).

blackbit42, 2024-12-12:
[image]

I tried the same material color settings as you have them.
I must be doing something stupid...

> and didn't reset the scaling to 1

Where do I do that? FreeCAD? KiCAD?
What I have found is this, but it is (and was) set to 1.

[image]

30350n, 2024-12-12:
> I tried the same material color settings as you have them.

Hmm that's odd. And looking at your screenshot, the colors **do** seem to appear in KiCAD, right?
Any errors when importing the model in Blender? Oh actually: Are you sure you are using the modified VRML importer from `pcb2blender`? `Import -> X3D/VRML (.x3d/.wrl) (for pcb3d)`

KiCAD scaling looks correct. Is the scaling in FreeCAD correct too? How big is the relay in reality?

blackbit42, 2024-12-12:
[image]

Yes, in KiCAD I see color.
Yes, I am using the correct import dialog in Blender.
There are no messages on the shell where I've started Blender at all.
[image]
These are the dimensions from the official datasheet.
35.24 * 9.9 * 9.9 mm

30350n, 2024-12-13:
Very odd. Can you share the file? Also, just to make 100%, these .wrl files are exported using `free2ki`? Because by default `free2ki` actually exports compressed `.wrz` files. Maybe that's the issue actually?

blackbit42, 2024-12-13:
[image]

That was indeed the issue. I had used the export function of FreeCAD itself, not of `free2ki`.
Now, everything fell into place. Wonderful!

[image]

This is only a 'Workbench' render. The other 2 engines would take 8 hours or so. I am on a laptop with intel graphics...

I think that's it.
Thanks so much for your patient help with this, much appreciated.
Would you be interested to tinker with the viewports thing between the three programs?
Perhaps on an IRC or Discord channel where you are?

closed 2024-12-13 by blackbit42

30350n, 2024-12-13:
Okay perfect, that explains it.

> Would you be interested to tinker with the viewports thing between the three programs?
> Perhaps on an IRC or Discord channel where you are?

I probably won't be able to spend much time on looking into this myself, but if you want to update me about your progress/findings or any issues you are running into, I'd be happy do communicate via discord DMs (my userid is `30350n`) ^^