30350n/pcb2blender #87: Olimex export/import problem - FR4 color error while exporting

goran-mahovlic, 2024-12-13:
Hi, it is me again.

I am trying to get Olimex board exported to pcb3d, and getting this error

[image]

This error can be fixed with removing FR4 color in stack up settings.

If I remove FR4 color, and try to import pcb3d into blender, blender 4.2 just closes ( I have just 8GB of RAM so maybe that is reason. )

Could you please check?

Olimex board is here:

https://github.com/OLIMEX/GateMateA1-EVB/tree/main/HARDWARE/GateMateA1-EVB-Rev.B

30350n, 2024-12-13:
Related to #68.

> This error can be fixed with removing FR4 color in stack up settings.

Yeah, either do that or actually specify colors for the top/bottom Mask and Silkscreen layers as a workaround.

> If I remove FR4 color, and try to import pcb3d into blender, blender 4.2 just closes ( I have just 16GB of RAM so maybe that is reason. )
>
>Could you please check?

Can you share the `.pcb3d` file? If you suspect it's the complexity of the file (which seems reasonable), you can also try importing without components in Blender. My best guess is that this (similarily to [#85](https://github.com/30350n/pcb2blender/issues/85#issuecomment-2535907773) is being caused by overly detailed custom 3d models.

goran-mahovlic, 2024-12-13:
I was able to import into blender 4.2 with removing merge materials 4.3 still exits in my case.

[image]

I also imported all other extensions and it is looking really nice!

[image]

Here is pcb3d file

https://wetransfer.com/downloads/87992c457ec0f959484bf94a4ec7be4320241213170022/9fa09825661f3bfb0734f2ca2518302220241213170048/482852?t_exp=1734368422&t_lsid=d4a4e590-9990-4b24-a934-0f45352e9b8c&t_network=email&t_rid=ZW1haWx8Njc1YzY4MjZkODI3Zjg0MWYyMzU0MThj&t_s=download_link&t_ts=1734109248&utm_campaign=TRN_TDL_01&utm_source=sendgrid&utm_medium=email&trk=TRN_TDL_01

30350n, 2024-12-13:
Very nice render you created there!

I am able to reproduce the crash. Seems to be caused by Blender's boolean code.

[image]

So the `Merge Materials` setting isn't the caveat here, but the `Cut PCBs` one is. Disabling it makes the import pass.
You can also just not define the board outlines, if your KiCAD PCB file only contains a single board:

[image]

30350n, 2024-12-13:
(The exporter issue will be fixed in the next release).

goran-mahovlic, 2024-12-13:
Perfect! Thank you!

30350n, 2025-04-12:
This should be fixed in [v2.17](https://github.com/30350n/pcb2blender/releases/tag/v2.17-k9.0-b4.2lts).

closed 2025-04-12 by 30350n