30350n/pcb2blender #88: Fedora 41 - python 3.13 - can't export

the-quix, 2024-12-17:
Traceback (most recent call last):

  File "/home/cicero/.local/share/kicad/8.99/3rdparty/plugins/com_github_30350n_pcb2blender/__init__.py", line 22, in Run
    export_pcb3d(dialog.file_picker.GetPath(), boarddefs)
    ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/home/cicero/.local/share/kicad/8.99/3rdparty/plugins/com_github_30350n_pcb2blender/export.py", line 83, in export_pcb3d
    pcbnew.ExportVRML(wrl_path, 0.001, True, True, components_path, 0.0, 0.0)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

TypeError: ExportVRML() missing 2 required positional arguments: 'aXRef' and 'aYRef'

30350n, 2024-12-17:
You are using a prerelease version of KiCad which is not supported (this is a duplicate of #75).

hfieger, 2025-03-03:
I have the same problem on WIN11 24H2 using the current version Kicad 9.0.

30350n, 2025-03-03:
There's no release that supports KiCad 9.0 yet.

hfieger, 2025-03-03:
Thanks for answering so quick.

I thought it's alreday in:
https://github.com/30350n/pcb2blender/tree/master/pcb2blender_exporter

Any idea when?

30350n, 2025-03-03:
The fix for this issue is there already, but I haven't found time to actually test if that's enough to make everything work on 9.0 or if there are any issues yet. So there's no release. You can always manually download the `master` version and install it.

closed 2025-05-24 by 30350n