30350n/pcb2blender #89: Kicad 9 support

KOEGlike, 2025-01-03:
rc1 has released, so this is almost the final version, and the new API shouldn't change any more

KOEGlike, 2025-01-03:
The API change broke the KiCad exporter, `ExportVRML_File()`  has now two new parameters, `aXRef` and `aYRef`, [(docs)](https://docs.kicad.org/doxygen-python-nightly/classpcbnew_1_1EXPORTER__VRML.html), if you try to

closed 2025-01-06 by 30350n

marcdraco, 2025-03-06:
Still broken in KiCAD 9 for me. I don't really understand enough about Python to check the version number.

I'm on FlatPak so I'm a bit stuck if the KiCAD plugin installer doesn't deliver the correct version.

Any ideas when this will be pushed or  is there a manual way I can install the latest one? Being >60 makes learning new stuff a lot harder than I would ever have guessed.

30350n, 2025-03-06:
You can manually download/clone the repository and copy the `pcb2blender_exporter` folder to your KiCad plugins directory.

marcdraco, 2025-03-08:
Thanks Bobbe, I'll try that. I hate it when excellent plugins like this one get busted when small things are changed in the host app. I thought I was going bonkers again when it started failing on me! :)

KiCAD Release 9 has had some other changes that have caught me with my (metaphorical) pants down. I'm looking forward to making an announcement video with Blender and this is perfect.