30350n/pcb2blender #93: STEP components model

bojczmac, 2025-01-19:
Doesn't work if PCB contains components model in step format.

30350n, 2025-01-19:
The exporter should generally handle that. Can you provide more detail into the issue you're experiencing?

bojczmac, 2025-01-19:
You have right. That was my fault. But I see that footprint with thermal via's is not display properly in blender. 

[image]

30350n, 2025-01-19:
Solder joints will never be created for vias. I'm guessing those are THT pads? Is the footprint they are part of marked as SMD? If so, I guess solder joints for THT pads of SMD components should maybe not be created.

bojczmac, 2025-01-19:
Yes, these are THT pads with fabrication property set to Heatsink pad.

30350n, 2025-01-19:
Ah, that's perfect I should be able to filter for that, thanks!

30350n, 2025-04-12:
This should be resolved with [`24ce234`](https://github.com/30350n/pcb2blender/commit/24ce234304a73941d1df5b9377f9a2231da4f41c) in [v2.17](https://github.com/30350n/pcb2blender/releases/tag/v2.17-k9.0-b4.2lts). Solder joints will only be created for pads with Fabrication Properties set to 'NONE', 'BGA' or 'CASTELLATED'.

closed 2025-04-12 by 30350n