30350n/pcb2blender #97: Does not work in kicad 9

picli3, 2025-03-05:
I've been using your plugin for a long time, thank you very much!!

But since I installed kicad 9, it doesn't work, can you help me with this?

[image]

30350n, 2025-03-05:
There is no release for KiCad 9.0 yet. This is a duplicated of #89.

closed 2025-03-05 by 30350n

marcdraco, 2025-03-06:
I'm getting the same error using the plugin installed from KiCAD 9.0 which got bumped by my FlatPak's autoupdate.

Any idea when the new version will drop?

30350n, 2025-04-13:
[v2.17](https://github.com/30350n/pcb2blender/releases/tag/v2.17-k9.0-b4.2lts) supports KiCad 9.0 now. It should also be available via the internal Plugin and Content Manager in the next few days.

picli3, 2025-04-13:
Thanks for the update, although now I have a problem installing the new version in Blender 4.1.1

[image]

30350n, 2025-04-13:
Blender 4.1 is not supported anymore. Blender 4.2 introduced a new [Extension Format](https://extensions.blender.org/about/) which is incompatible with the old addon format.

pcb2blender supports the newest Blender LTS version (4.2 atm) and the newest non LTS version (4.4).

picli3, 2025-04-13:
It works now, thank you very much!
[image]