30350n/pcb2blender #98: "Incompatible" error in KiCAD 9

marcdraco, 2025-03-09:
I'm on Debian 12 which entails a FlatPak - and this is causing me issues all over the place (Blender, also a FlatPak, is similarly affected).

Specifically when I try to use the manual import I get an error:

[image]

Neither exiting nor helpful to me or the developer?!

FlatPak is great until programs need to access extensions it appears.

30350n, 2025-03-10:
If you installed the version from the `master` branch this should not happen. The `kicad_version` got bumped to `9.0` in #90.

Also not sure what you mean with "manual" import. To manually install the exporter from the `master` branch, download or clone the repository, then copy the `pcb2blender_exporter` folder to your [KiCad Plugin Directory](https://dev-docs.kicad.org/en/apis-and-binding/pcbnew/):

[image]

aperezbios, 2025-03-10:
@KOEGlike can we pretty-please get a release of pcb2blender that supports KiCad 9? It's been more than two weeks since the 9.0.0 release, and the issue was resolved months ago. All we need is a release.

marcdraco, 2025-03-10:
I'm convinced this is an issue with ****Ing FlatPak. 

If you have a spare machine (or you can use a box) to drop a copy of Debian 12 in, you'll see it's a FlatPak. Debian Sid has the 9.0 binaries as a backport but I'm just not confident enough to use the unstable code in case it breaks something else, if that makes sense.

The directories you've listed (bang on for a "normal" install) aren't present or used in the Flatpak. I don't know about Snap as that's more of an Ubuntu thing. I have a couple of Windows machines so I'm not totally cornered but the laptop gets very hot on my lap and the tower is louuuuud. They're both mid-level gaming/rendering boxes which explains it.

Blender seems to be similarly hobbled by FlatPak - I don't know if they consider this a bug, a feature or is even by design for the isolation. FlatSeal (which is pretty much an essential for peons like me) gets around some of the restrictions - I store all of my work on a local server which is normally blocked by the security). It won't fix this though.

As FlatPak is used more extensively, since the isolation is so convenient and they can bundle versions way ahead of the binaries from the distro (V7 on Debian 12).

I can't even export as VRML, etc. from PCBNew because all the textures go missing.  It's not fair on the devs like but if you can get to the bottom of it, I'll happily share this with the KiCAD/FlatPak team as a bug.

marcdraco, 2025-03-10:
> [@KOEGlike](https://github.com/KOEGlike) can we pretty-please get a release of pcb2blender that supports KiCad 9? It's been more than two weeks since the 9.0.0 release, and the issue was resolved months ago. All we need is a release.

The mainline ZIP should work well so long as you're not in a FlatPak version. If you are, we're in the same boat... I wish PCBNew could export to earlier versions even if it loses editing features like those lovely Beziers.

30350n, 2025-03-10:
> The directories you've listed (bang on for a "normal" install) aren't present or used in the Flatpak.

Try the directory you get from "Open Plugin Directory" in the PCB Editor:

[image]

marcdraco, 2025-03-10:
I tried every exporter they have (at least ones I can import into Blender) - I had been able to get that to work before. My excuse is I'm ancient and I'm sticking to it. I've found the directory under /home/marc/.var/(etc.)

I'll give that a go now and get right back to you. I was just trying the Ubuntu machine... :)

30350n, 2025-03-10:
[image]

This part of the path looks weird, but the folder name shouldn't actually matter. What's more important is that you didn't download the [repository from the `master` branch](https://github.com/30350n/pcb2blender/archive/refs/heads/master.zip).

Paschalis, 2025-03-10:
## Incorrect Parameter Order in `pcbnew.ExportVRML` Causes Missing `aXRef`/`aYRef` Error

### Description  
The current `ExportVRML` call in the PCB3D export script throws a `TypeError` due to parameter misalignment. This issue affects users who installed the plugin via KiCad's Plugin/Content Manager. According to the [EXPORTER_VRML Class Reference](https://docs.kicad.org/doxygen/classEXPORTER__VRML.html), the method requires parameters in this specific order:

```cpp
bool ExportVRML_File(
    PROJECT* aProject,             // [Internal] Not needed in Python
    wxString* aMessages,           // [Internal] Not needed in Python
    const wxString& aFullFileName, // Output .wrl path
    double aMMtoWRMLunit,          // mm-to-VRML-unit conversion
    bool aIncludeUnspecified,      // Include unspecified components
    bool aIncludeDNP,              // Include DNP components
    bool aExport3DFiles,           // Export 3D model files
    bool aUseRelativePaths,        // Use relative paths
    const wxString& a3D_Subdir,    // 3D model subdirectory
    double aXRef,                  // X reference offset
    double aYRef                   // Y reference offset
)
```

## Original Broken Code
```python
# From plugin version shipped via KiCad Content Manager
pcbnew.ExportVRML(wrl_path, 0.001, True, True, components_path, 0.0, 0.0)
```

## Error Message
```
TypeError: ExportVRML() missing 2 required positional arguments: 'aXRef' and 'aYRef'
```

## Verified Solution
As resolved in https://github.com/30350n/pcb2blender/issues/89

```python
pcbnew.ExportVRML(
    wrl_path,               # Output .wrl path
    0.001,                  # mm-to-VRML-unit (0.001 = meters)
    True,                   # aIncludeUnspecified
    True,                   # aIncludeDNP
    True,                   # aExport3DFiles (export 3D models)
    True,                   # aUseRelativePaths (relative paths)
    str(components_path),   # a3D_Subdir (explicit string conversion)
    0.0,                    # aXRef (X offset)
    0.0                     # aYRef (Y offset)
)
```

marcdraco, 2025-04-02:
Oh mate, thank you! I missed that - would you imagine, GitHub had hidden it as SPAM!

closed 2025-04-02 by marcdraco