30350n/pcb2blender #99: Error when exporting PCB with KiCAD 9.0

a-me-lia, 2025-03-15:
[image]

Windows 10 22H2 fresh install

PoleG97, 2025-03-19:
Make a manual install of `main` branch.

Steps for manual install are covered in `README.md`.

harold42343253259023592375420374027340, 2025-03-25:
I am getting the same issue, but even with a manual install i still get the issue
Kicad 9.0

repsej-h, 2025-04-02:
Hey 

I am also experiencing this issue, both with manual installation and installation using the plugin manger. I'm also using KiCad 9.0

The project can be found here: https://github.com/repsej-h/photogate

30350n, 2025-04-04:
Duplicate of #89.

closed 2025-04-04 by 30350n

30350n, 2025-04-13:
[v2.17](https://github.com/30350n/pcb2blender/releases/tag/v2.17-k9.0-b4.2lts) supports KiCad 9.0 now. It should also be available via the internal Plugin and Content Manager in the next few days.