30350n/pcb2blender #100: Error at time to install blender importer in blender 4.4.0

JairoGPR, 2025-03-19:
I can't install the extension from blender marketplace.

[image]

And it shows this error

`pcb3d_importer: This Blender version (4.4.0) must be less than the maximum version (4.4.0)`

[image]

ziomau10, 2025-04-01:
I have the same problem.
I tried to manually change version number from 4.4.0 to 4.4.9 in blender_manifest.toml, but then I get the error:
```
Exception in module register(): C:\Users\mau\AppData\Roaming\Blender Foundation\Blender\4.4\extensions\user_default\pcb3d_importer\__init__.py
Error: No module named 'error_helper'
```

30350n, 2025-04-04:
Blender 4.4 is not supported yet.

30350n, 2025-04-12:
The latest release ([v2.17](https://github.com/30350n/pcb2blender/releases/tag/v2.17-k9.0-b4.2lts)) supports Blender 4.4 now.

closed 2025-04-12 by 30350n