30350n/pcb2blender #101: Crash exporting from Kicad V9.0.x

ziomau10, 2025-03-30:
As soon as I try to export, i Get:

[image]
I have placed the two text PCB3D_TL_Main and PCB2D_BL_Main in the board corners

I am using a recent RC for V9.0.1, but previously I had the same error with official V9.0.0 release

Should you need further information, logs or anything else, please ask.

Great Job.
Mau.

ziomau10, 2025-03-31:
Digging a little bit, it looks like in V9.0, the function ExportVrml requires a couple of additional parameters:
`def ExportVRML(self, aProject, aMessages, aFullFileName, aMMtoWRMLunit, aIncludeUnspecified, aIncludeDNP, aExport3DFiles, aUseRelativePaths, a3D_Subdir, aXRef, aYRef)`
alas the documentation still reports:
`def pcbnew.ExportVRML(aFullFileName, aMMtoWRMLunit, aIncludeUnspecified, aIncludeDNP, aExport3DFiles, aUseRelativePaths, a3D_Subdir, aXRef, aYRef)`

Hope this helps ...
Mau.

DimitriOnLSD, 2025-03-31:
This has already been fixed in 0aa190c.

Search for the file "...Documents\KiCad\9.0\3rdparty\plugins\com_github_30350n_pcb2blender\export.py" and replace it with [`export.py`](https://github.com/30350n/pcb2blender/blob/master/pcb2blender_exporter/export.py) file.

Or change the line 83 of that file to:
```
pcbnew.ExportVRML(wrl_path, 0.001, True, False, True, True, components_path, 0.0, 0.0)
```

christianhalter, 2025-04-04:
> This has already been fixed in [0aa190c](https://github.com/30350n/pcb2blender/commit/0aa190cbc60c411c032aef5693da420b42a06f2a).
>
> Search for the file [...]

It's not fixed. I tried your workaround and now I get this:

[image]

I think @ziomau10 is right, I've also found this [Deprecation notice in Kicad's dev docs](https://dev-docs.kicad.org/en/apis-and-binding/pcbnew/index.html#_deprecation_notice), so I guess most plugins should be updated.

30350n, 2025-04-04:
> It's not fixed. I tried your workaround and now I get this:

It is. You just didn't restart KiCad. The fix is also already present on the `master` branch, but there's no release yet because I'm still working on some things.

> I've also found this [Deprecation notice in Kicad's dev docs](https://dev-docs.kicad.org/en/apis-and-binding/pcbnew/index.html#_deprecation_notice), so I guess most plugins should be updated.

KiCad 10 is still almost a year away, but more importantly, afaik, the new IPC API doesn't support some functionality this addon relies on, namely:
> The IPC API currently has no support for plotting or exporting files from KiCad designs.

(Taken from [here](https://dev-docs.kicad.org/en/apis-and-binding/ipc-api/for-addon-developers/)).

30350n, 2025-04-04:
Duplicate of #89.

closed 2025-04-04 by 30350n

30350n, 2025-04-13:
[v2.17](https://github.com/30350n/pcb2blender/releases/tag/v2.17-k9.0-b4.2lts) supports KiCad 9.0 now. It should also be available via the internal Plugin and Content Manager in the next few days.