30350n/pcb2blender #105: Windows ARM64 support

SteynGuelen, 2025-04-16:
Hi, I'm using a Surface Laptop 7 which has a Qualcomm ARM processor/gpu.

When installing in Blender, I get this error:
pcb3d_importer: This platform (windows-arm64) isn't one of (windows-x64, linux-x64, macos-x64, macos-arm64)

Is ARM64 support on the roadmap?

30350n, 2025-04-18:
The blocker here is [skia](https://pypi.org/project/skia-python/#description) not supporting (or atleast not offering pre-built wheels) for win-arm64.

30350n, 2025-04-18:
Related: [skia-python#318](https://github.com/kyamagu/skia-python/issues/318)

HinTak, 2025-04-18:
Hmm, a little surprised that somebody wants this, but I am adding win11 arm builds to https://github.com/kyamagu/skia-python/pull/314 . Give that a try when the CI wheel build finishes.

HinTak, 2025-04-18:
There is a good chance that I don't want to do arms  windows skia-python m87 (skia m87 itself didn't support arm linux nor arm windows; and doing skia python m87 for arm linux has been painful), so I'd like you guys to try moving to the m1xx to see if you are missing anything.

30350n, 2025-04-18:
Cool, thanks a lot for that! What's the reason for all the 1xx builds being marked as beta/prereleases?

HinTak, 2025-04-18:
It looks like we (skia-python) are dependent on numpy being more available on win on arm : https://github.com/numpy/numpy/issues/22530#issuecomment-2816320842 . As for the beta/prerelease status, it is mostly that some portion of m87 functionality was moved 
https://github.com/kyamagu/skia-python/issues/304 - and we haven't figured out what the corresponding replacements are, or even whether they should be provided / anybody needing them. Historically, skia-python m87 was quite exhaustive - that is a good thing as it covers a large portion of skia,  but a bad thing when you realise about 20% of API changes between m87 and m116!

HinTak, 2025-04-18:
I think with m134/m136 we cover about 93% of m87 - but the remaining is hard, and we don't know if anybody need/want the remaining.

30350n, 2025-04-19:
I'll take a look at it sometime next week. I'm really only using skia in one place to rasterize some svg files, so we should probably be fine.

HinTak, 2025-04-19:
The svg module went non-experimental in m88, and got some improvement / enhancement since; so this is definitely an area where you'd benefit. There is an outstanding svg-related enhancement (https://github.com/kyamagu/skia-python/issues/307) which might worth you commissioning/ sponsoring me to get it done 😉

HinTak, 2025-04-23:
Aparently python being x64 on arm windows is known https://github.com/actions/partner-runner-images/issues/85 - but 3.11 / 12, 13 native are available for arm windows.

HinTak, 2025-05-01:
I have submitted https://github.com/numpy/numpy/pull/28836 - and AFAIK we are dependent on that (or something similar / equivalent) being merged into a numpy release before skia-python can add arm windows support.

HinTak, 2025-05-01:
Btw, if somebody is motivated enough, it should ne possible to use https://github.com/HinTak/numpy/actions/runs/14644630041 or https://github.com/numpy/numpy/pull/28836 
 in addition to https://github.com/kyamagu/skia-python/pull/314 to build arm windows skia-python . It is just not feasible distributing a bundled numpy, so we are basically waiting for the merge and new release of numpy.

HinTak, 2025-05-20:
Apparently, numpy has a half-yearly release schedule. As my pull got merged a few weeks ago, we are waiting for numpy 2.3.0 (possibly with a rc1 before it) late June early July. We have skia-python 138 next week, then early July should be 140.

HinTak, 2025-05-26:
Numpy 2.3.0rc1 is out, so I think I have done the last of https://github.com/kyamagu/skia-python/pull/314 and will merge that as skia-python v138.0rc1, in about a week's time.

Note that apparently opengl is not working on arm windows - that would be the angle build, another item in skia-python 's tracker. Does pcb2blender uses opengl via skia?

HinTak, 2025-05-31:
Just FYI, skia-python 138rc1 will be out soon. Apparently opengl should work on arm windows too with a compatibility pack from microsoft. So we have enabled it. I'd be interested to know / confirm that opengl works on ARM64 windows and some details/steps.

HinTak, 2025-06-04:
@30350n I have tagged v87.9 rc1 to build and it should be uploaded to pypi in about an hour. You need --pre to get it (together with numpy 2.3.0 rc1 too). v138.0rc1 should be out later this evening too... I would prefer you to use 138rc1; 87.9rc1 is for people who can't upgrade (there is still a small percentage of incompatibility between 87 and 138).

v87.10 will be the last of v87.x .

HinTak, 2025-06-04:
v138.0rc1 (and 87.9rc1) have now  both landed in pypi.

HinTak, 2025-06-10:
Numpy 2.3.0 proper was out, so skia-python 138 (and 87.9) is out with windows arm64 wheels.

30350n, 2025-09-10:
I added Windows arm64 support in [v2.17.3](https://github.com/30350n/pcb2blender/releases/tag/v2.17.3-k9.0-b4.2lts) now and also updated to skia-python 138. The dependency on libegl1 is bugging me a little bit, but let's see if people figure it out.

Thanks a lot for your work on supporting the arm platforms in skia-python @HinTak ! And sorry for not answering earlier here ^^