30350n/pcb2blender #107: New version no longer compatible with Blender Add-On

celeronpm, 2025-04-25:
I am using the [blender PCB 3D Importer](https://extensions.blender.org/add-ons/pcb3d-importer/?utm_source=blender-4.3.2) and the new version exports files which are not compatible.

Blender gives this error when trying to import the file

`Python: Traceback (most recent call last):
  File "C:\Users\celer\AppData\Roaming\Blender Foundation\Blender\4.3\extensions\blender_org\pcb3d_importer\importer.py", line 321, in import_pcb3d
    pcb = self.parse_pcb3d(file, tempdir)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\celer\AppData\Roaming\Blender Foundation\Blender\4.3\extensions\blender_org\pcb3d_importer\importer.py", line 684, in parse_pcb3d
    layers_bounds = struct.unpack("!ffff", layers_bounds_path.read_bytes())
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Blender Foundation\Blender 4.3\4.3\python\Lib\zipfile.py", line 2472, in read_bytes
    with self.open('rb') as strm:
         ^^^^^^^^^^^^^^^
  File "C:\Program Files\Blender Foundation\Blender 4.3\4.3\python\Lib\zipfile.py", line 2433, in open
    raise FileNotFoundError(self)
FileNotFoundError: D:\User\Desktop\lf-audio-hardware\HPVM\hpvm.pcb3d/layers/bounds
`

30350n, 2025-04-25:
I did some file format changes in the latest version. Updating to the newest importer extension should fix this.

celeronpm, 2025-04-25:
OK, I see the issue, my blender isn't updating my add-on. I'll figure it out.

[image]

closed 2025-04-25 by celeronpm