30350n/pcb2blender #109: Supporting linux aarch64

xeniagda, 2025-04-29:
Hi! Similarly to https://github.com/30350n/pcb2blender/issues/105 I would love releases for aarch64/arm linux. In that issue, skia not having builds for Windows ARM was the blocker, but it seems here that skia should have builds for aarch64 linux (skia_python-87.8-cp313-cp313-manylinux_2_28_aarch64.whl)

Is there anything else blocking pcb2blender?

xeniagda, 2025-04-29:
Oh I didn't know that, using nixos it's built from source in nixpkgs and works on linux-aarch64 without issues. But since the builder runs on ubuntu, it is a bit problematic to run blender for aarch64 there...

xeniagda, 2025-04-29:
I can build the plugin successfully on nixos (with some minor patches). Seems cross-building works from aarch64 to other platforms (although I haven't tested the plugin does run). Would it work to cross-build to linux aarch64 on the ubuntu x86 host platform? Should be as easy as adding `"linux-arm64": ("manylinux_2_28_aarch64", "manylinux_2_17_aarch64"),` to `BLENDER_ADDON_SUPPORTED_PLATFORMS` in `download_dependencies.py`, along with adding `linux-arm64` to `platforms` in `pcb2blender_importer/blender_manifest.toml`.

30350n, 2025-05-01:
This actually required some more refactoring in `download_dependencies.py`, because of version mismatches between `skia-python` and `numpy`, but I got a build now. Can you check if this works for you?

[pcb2blender_importer_v2-17-2_b4-2.zip](https://github.com/30350n/pcb2blender/releases/download/v2.17.2/pcb2blender_importer_v2-17-2_b4-2.zip)

HinTak, 2025-05-01:
There is no need to cross-build for aarch64 linux since a couple of months ago. We are switching to native aarch 64 linux build for the next release: 
 https://github.com/kyamagu/skia-python/pull/314 (ignore the CI failure - it passed until I added arm windows suppport to that pull - I 'll remove those commit if numpy arm windows isn't ready by next release, before merging). Also current is v136.b13 - v87.x is dead end and has been dead end for about 4 years...

mannireis, 2025-09-10:
cool thx