30350n/pcb2blender #111: Solder pads don't have solders

tapegoji, 2025-05-19:
Hi, this might be a feature request or simply not. 

I am looking to see the solder joints of the smd in blender. I don't see them. Also, threre is a tool that can model solder joints and we can integrate it into this git hub. take a look at this link 

[https://kenbrakke.com/evolver/evolver.html](url)

30350n, 2025-05-19:
SMD solder joints get created for almost all Pads, they are just not that big/visible.

Currently the meshes for all solder joints are basically "procedurally generated" based on the Pad data, but don't take the actual footprint 3D models into account.

I agree that especially for SMD Pads they could be improved, but I'm pretty sure you could already get big improvements by tweaking the currently used generation code a bit.
I've also thought about using [Blender's Metaballs](https://docs.blender.org/manual/en/latest/modeling/metas/introduction.html) for this, but decided against it because of the added complexity when I first implemented the feature.
Using the external tool you linked, would probably be way out of scope for this, as integrating an external tool/library like that into a Blender extension is not trivial at all and would require **a ton** of work. So if you want to look into this, I'd recommend going with one of the other options.

No1089, 2026-11-02:
Following up on your point that the current joints don't take the footprint 3D models into account: I ran into the same thing rendering my own boards from KiCad in Blender and ended up writing joints that do. It's a standalone module that might be useful here: https://github.com/No1089/kicad-blender-render/tree/main/solder_joints (LGPL-3.0, so it can go into a GPL-3 project like this one).

Instead of a template mesh per pad type, it ray-casts the component models above each pad and builds one joint per pad:

- **SMD:** concave meniscus from the terminal down to the pad edge. Solder only climbs metal (end caps, leads, plated flanks), never plastic. Leadless parts (QFN/DFN/LGA) get the standoff filled and a small toe fillet.
- **THT:** the hole is probed along its axis, so the real pin cross-section (round pin, flat blade, offset lead) shapes the joint. Large fillet on the solder side, small one on the component side.
- **HASL:** optional thin tin coating on bare pads and empty holes.

**No external tools:** plain Python with NumPy and `bpy`, no compiled code or extra installs, so it can sit in the extension as one folder. The core is NumPy only and has tests that run without Blender; a thin Blender layer adds the BVH prober, mesh building and a solder material.

**How it fits pcb2blender:** `pad_outline(shape, size, position, rotation, roundness)` takes the pad data the exporter already writes to the `.pcb3d` (position, side, has_paste, type, shape, size, orientation, roundness, drill), and the README has an integration sketch.

**Renders:** https://github.com/No1089/kicad-blender-render#gallery (including a with/without solder comparison).

Happy to open a PR adding it as an optional "model-aware" choice next to *Add Solder Joints*, or to reshape it to suit the project.