30350n/pcb2blender #113: Blender Python3.13 support

MTRNord, 2025-05-23:
Hi :)

It seems like I cant use this with fedora currently as I am getting this error:

> RuntimeError: Extension bl_ext.blender_org.pcb3d_importer is incompatible (This Python version (3.13) isn't compatible with (3.11))

any ideas if there is a workaround? I am on fedora 42 which bundles 3.13. I tried to use pyenv via setting the pythonpath which did however not work.

30350n, 2025-05-23:
This is a duplicate of #76 . The problem here is that your linux distribution is shipping a broken Blender package (using a Python version which is not 3.11). **Please complain to your package maintainers about this.**

You can either use something like flatpak to install a working Blender version or alternatively try to bump the versions in this extension yourself and build your own version.

closed 2025-05-23 by 30350n

set-soft, 2025-05-24:
> No. Imo, the whole idea of only having a single version of a software/software dependency installed (or frozen by a packaging repository) has completely fallen out of time. There's really not many good reasons for it at all, except for the possible disk/memory savings (which, are negligible on most modern systems).

We have a radical different point of view. Just note that Blender/Python way of doing things is aligned with Windows, Linux is aligned with UNIX style. They are radically different.

30350n, 2025-05-24:
Hm, I don't know. I don't really see this as a Windows vs. Linux thing tbh. Like I said, just take a look at how NixOS handles things.

The problem here really is package repositories beating a dead horse, trying to somehow make all their packages compatible to eachother while sticking to a specific version for all of them.

set-soft, 2025-05-24:
> The problem here really is package repositories beating a dead horse, trying to somehow make all their packages compatible to eachother while sticking to a specific version for all of them.

Which saves:
- RAM, one instance in memory at a time, not N versions of the same lib
- Storage, for similar reason
- Download bandwidth
- Headaches, because you remove a lot of reproducibility issues. You know the versions are consistent.

Yes it takes more time from developers, but saves time and resources to users. The effort of one developer, reduces the effort of N users.

I don't know the details of NixOS, but from a quick look at its main page it seems to address only one of the things I mentioned.
And note that you can offer users the choice:

1. Waste resources but make things "simpler"
2. Don't waste resources

In my case I offer docker images, suitable for CI/CD, but also allowing simple use, very inefficient.
But I also offer Debian packages, so you can install the tools in very efficient way.
Is not strange that the most problematic thing is Blender.

30350n, 2025-05-24:
> I don't know the details of NixOS, but from a quick look at its main page it seems to address only one of the things I mentioned.

It actually addresses [all of them](https://nixos.org/guides/how-nix-works/). It simply allows each package to work with the specific dependency versions it requires, while still reusing dependencies that are required by multiple different packages. This results in your 'N' being kept small, while eliminating all the discussed versioning issues and ensuring reproducibility to a level no other system offers.

> Yes it takes more time from developers, but saves time and resources to users. The effort of one developer, reduces the effort of N users.

I still just have to disagree on this stance that this is something developers should even remotely care about. If I as a developer create a program, I simply define what dependencies versions it requires and that's it.
If some package maintainer now decides to add **my program** to some package repository, it's obviously **THEIR** responsibility to ensure that the package is actually working correctly.